timschmidt/repsnapper #121: Lubuntu in 64bits and Prusa i3

jesusansal, 2016-10-30:
Hello
 First of all sorry if my English is not 100% correct.
Second, I installed repsnapper I connected after load the marlin by arduino software... I printed a few squares to configurate the setting .... and I started to feel happy with my progress but.... I take a step forward I try to print a cilinder... and the result are not very good:
1º the noozeld (or how ever spell it) is starting to extrude some pla after the 4 layer so no posibility to stick to the plataform 
2º the Z Axis is not moving anymore... 

Thanks I have been working before with Pronterface and Cura I will go back to that software.

jesusansal, 2016-11-01:
I fix it, the Voltage of the motors was not enoght... now work perfect... One question If someone can help.... how can I use the function Post porcessor Gcode ?

hurzl, 2016-11-02:
I'm not sure what you mean by "Post porcessor Gcode"